ztachip/ztachip #1: couldnt find many documentation and open communiction channels

Harshil-C, 2022-09-27:
hey, iam an engineering student and is  new to this project and want to know more about this,can i use ther fpga boards other the recommended one and can  get link for the slack channel

ztachip, 2022-09-27:
Hi Harshil

ztachip can be compiled into any FPGA or even ASIC.
the only source you need are:

  *   All files under https://github.com/ztachip/ztachip/tree/master/HW/src
  *   They are generic VHDL codes without any special primitives so it is ready to be ported to any FPGA

  *   Have a version of wrapper library for your FPGA/ASIC
  *   https://github.com/ztachip/ztachip/tree/master/HW/platform
  *   There are 4 components that you need to map to your FPGA library. They are just some basic memory block primitives so any FPGA/ASIC would have them.

Then you can look at https://github.com/ztachip/ztachip/blob/master/HW/examples/GHRD/main.v as example on how to integrated ztachip to your code.

  *

Please check back https://github.com/ztachip/ztachip . I should have more documentation by next month.

Regards
Vuong

closed 2022-10-01 by ztachip

ztachip, 2022-10-01:
I have added a section on main page on how to port both hardware/software stack.
Let me know if I missed anything

bittu-digital, 2024-10-28:
can i have more detailed documentation on the project. From the replies and after looking into https://github.com/ztachip/ztachip/tree/master/HW/src , I understood how they are connected but I could not understand why they are connected and how each module works, how software part comes into the play where Ai models are deployed on the architecture. The project is interesting, and I would like to understand more about the project.

ztachip, 2024-10-29:
In ztachip github page, there are several documents links that might interest you

https://github.com/ztachip/ztachip/blob/master/Documentation/Overview.md
https://github.com/ztachip/ztachip/blob/master/Documentation/HardwareDesign.md
https://github.com/ztachip/ztachip/raw/master/Documentation/ztachip_programmer_guide.pdf
https://github.com/ztachip/ztachip/raw/master/Documentation/visionai_programmer_guide.pdf
https://github.com/ztachip/ztachip/blob/master/micropython/MicropythonUserGuide.md