ztachip/ztachip #2: compilation failed "final link failed: bad value"

kappamaster71, 2022-09-30:
Hi,
I tried building ztachip and got a final link failed error. I followed exactly the steps outlined, including fresh build of RISCV gcc. Here is part of the log:
`/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: warning: build/ztachip.elf has a LOAD segment with RWX permissions
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/lib/libstdc++.a(eh_globals.o): in function `_GLOBAL__sub_I___cxa_get_globals_fast':
eh_globals.cc:(.text.startup._GLOBAL__sub_I___cxa_get_globals_fast+0x0): undefined reference to `__dso_handle'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: eh_globals.cc:(.text.startup._GLOBAL__sub_I___cxa_get_globals_fast+0xc): undefined reference to `__dso_handle'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: build/ztachip.elf: hidden symbol `__dso_handle' isn't defined
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.2.0/../../../../riscv32-unknown-elf/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make: *** [makefile:118: build/ztachip.elf] Error 1`

Might you have any idea what's wrong? Thanks a lot.
Regards,

ztachip, 2022-10-01:
Please load latest version
You can see some explanation for the fix in ztachip/SW/src/main.cpp

closed 2022-10-08 by ztachip

kappamaster71, 2022-10-11:
thanks a lot, it's working now!