ztachip/ztachip #4: About Video Output

AugustZTR, 2023-03-09:
Hello, I compiled and wrote the program according to the document, and successfully used GDB for communication and program loading, however, I have not been able to get the output of the video screen, i'm not sure if it's the video output or if the program isn't working properly

ztachip, 2023-03-11:
Which FPGA board are you using?
If you are using the Arty-A7, can you try the project as is from github first? That should work fine.
If you create your own project, verify if your pin assignments for VGA are programmed correctly.

AugustZTR, 2023-03-12:
I'm using the same Arty-A7 100T as the project and still not figure out the reason.

ztachip, 2023-03-12:
But if you build ztachip project from git as is. Does it work?
Try to download latest ztachip again and open the project below to build
ztachip/HW/examples/GHRD/GHRD.xpr

AugustZTR, 2023-03-13:
I tried again and still not work.
The terminal output is like this
[image]

AugustZTR, 2023-03-13:
I tried again and still not work.
Here is the terminal output.
[image]
And here is the hardware.
[image]

AugustZTR, 2023-03-14:
Sorry, my bed. I mistook the following output as the "continue" command mentioned in the document
[image]
However, this has brought new problems. The board will reset after I input the "continue". I repeated the complete process on two computers respectively, and the result was the same. And still, no video output.
[image]

ztachip, 2023-03-14:
Looks like you have problem with the openOCD or USB connection.
Can you verify you setup correctly. The steps to setup this also on github page.
Make sure the board is operational before starting openOCD. Usually after power up, it takes 10sec before FPGA is programmed (you will see a green led turn on)

[??]<https://user-images.githubusercontent.com/28720897/224982776-fe67502e-3bee-4104-9b5a-be6ad2adaec2.png>
[??]<https://user-images.githubusercontent.com/28720897/224983449-fef2681d-8e33-4e3f-9b29-15b12deb6058.png>

ztachip, 2023-03-23:
Does it work for you?

ztachip, 2023-05-19:
Done

closed 2023-05-19 by ztachip