ztachip/ztachip #5: Error: 'JTAG scan chain interrogation failed: all zeroes' while launching OpenOCD

FaizanAhmad626, 2023-04-01:
Hey! I hope you're doing fine.
I am using this project with my FPGA device Genesys 2 Kintex-7 (XC7K325T-2FFG900C) and Vivado ML Edition 2022.1
Link to its reference manual : [https://digilent.com/reference/programmable-logic/genesys-2/reference-manual](url)

I did my best to change all the settings in IP cores and in main.v file for my device. I changed the DDR3 configuration, clock frequency etc. Then I changed its constraint file for my device. Instead of using Pmod VGA (because of unavailability) I used the built-in VGA connector (DB15) on mu FPGA device. I couldn't find anything to change in vga.vhdl but I changed the connections of VGA in constraint file. Although last following lines of the constraint file remain the same:
```
set_property CLOCK_DEDICATED_ROUTE FALSE [get_nets CAMERA_PCLK_IBUF]
set_property CLOCK_DEDICATED_ROUTE BACKBONE [get_nets clk_wiz_inst/inst/clk_in1_clk_wiz_0] # Added this line to avoid error in implementation
set_property C_CLK_INPUT_FREQ_HZ 300000000 [get_debug_cores dbg_hub]
set_property C_ENABLE_CLK_DIVIDER false [get_debug_cores dbg_hub]
set_property C_USER_SCAN_CHAIN 1 [get_debug_cores dbg_hub]
connect_debug_port dbg_hub/clk [get_nets clk]
```

I couldn't change these line because I couldn't understand them but they are linked with the debugging process I think.

Then I did Synthesis, Implementation and Generated Bit stream. Then I programmed the device through JTAG and using SPI flash and using the memory part found in its reference manual. The device programmed successfully after sometime as shown in the LED on my device.

Then I installed Ubuntu 22.04.1 in Oracle VM VirtualBox for the software part. I followed the instructions given in the readme.md document to install prerequisites, compiler for RISC-V and ztachip, OpenOCD and GDB Debugger.

But when I launch OpenOCD as instruction I encounter the following error, although OpenOCD still runs:
[image]
I didn't change anything in the configuration files being used. I couldn't understand them and I thought may be it'll work. If there is any change required in them for my device then kindly do let me know please.

Then I still launched GDB Debugger because OpenOCD was listening on port 3333 for GDB connections. When I entered monitor reset halt command it again gave me the same error:
[image]

After the program loaded the connection closed:
[image]

This is the view on the OpenOCD terminal:
[image]

Kindly help me out in this issue. Thank you!

ztachip, 2023-04-01:
Looks like you have a USB connection problem from VirtualBox
On VirtualBox configuration, go to Machine->Setting->USB then click on [+] button. You should see "Diligent USB device" option on the popup list. Then add the device. Then start your VirtualBox session.
Cheers
Vuong

[InkedScreenshot from 2023-04-01 19-39-41]<https://user-images.githubusercontent.com/111587195/229297629-cc65f858-5486-4c80-9179-2e50e40b0e1a.jpg>

[InkedScreenshot from 2023-04-01 19-42-46]<https://user-images.githubusercontent.com/111587195/229297837-9de5a09b-efa8-4258-8e77-d3a4ed9a46e9.jpg>

[InkedScreenshot from 2023-04-01 19-43-37]<https://user-images.githubusercontent.com/111587195/229297911-ec2f910f-c7c0-4ffc-95f5-a264f9ec6a94.jpg>

[InkedScreenshot from 2023-04-01 19-43-56]<https://user-images.githubusercontent.com/111587195/229297977-dc25406f-26e1-461e-a0cb-04b4d449dd47.jpg>

FaizanAhmad626, 2023-04-01:
Yes, I already did that while running this project before:
[image]

Also I updated the usb_connect.cfg file as follows:

```
if [info exists env(SPINAL_SIM)] {
    set SPINAL_SIM $::env(SPINAL_SIM)
} else {
    set SPINAL_SIM no
}

if {$SPINAL_SIM == "yes"} {
    interface jtag_tcp
    set _CHIPNAME fpga_spinal
    set TAP_NAME $_CHIPNAME.bridge
    set _CPUTAPID 0x10001fff
    jtag newtap $_CHIPNAME bridge -expected-id $_CPUTAPID -irlen 4 -ircapture 0x1 -irmask 0xF
    reset_config none
    adapter_khz 30000
} else {
    adapter driver ftdi
    ftdi device_desc "Digilent Adept USB Device"
    ftdi vid_pid 0x0403 0x6010
    ftdi channel 0
    ftdi layout_init 0x00e8 0x60eb
    ftdi tdo_sample_edge falling

    reset_config none
    adapter speed 30000

    source xilinx-xc7.cfg
    source jtagspi.cfg

    set TAP_NAME xc7.tap
}
```

i.e. changed 'Digilent USB Device' to 'Digilent Adept USB Device' to avoid error and some other changes which were recommended to me while running OpenOCD on terminal.

ztachip, 2023-04-01:
Unfortunately, I don't know much about openOCD to help you with this.
The configuration I use was an example that I downloaded from some websites.
But you can contact https://github.com/SpinalHDL/openocd_riscv for support.
I asked the author some questions before and he was very helpful.
He may already have an example for your board.
I hope this helps.
Cheers
Vuong

[Screenshot 2023-04-01 213210]<https://user-images.githubusercontent.com/111587195/229303664-78cb4c38-f2b9-4c1d-9249-591bb445235c.png>

FaizanAhmad626, 2023-04-01:
Okay. Thank you! :)

closed 2023-04-01 by FaizanAhmad626

FaizanAhmad626, 2023-04-01:
Just for the sake of information and knowledge.
Can you please explain those last lines of contraint file which I mentioned
in my first comment? What exactly those lines mean and their real purpose.
or any material where I can learn about them?
I wanted to ask about these configuration files also but unfortunately you
don't much know about them but thank you for referring someone else for
this.