ztachip/ztachip #7: Is it possible to increase the video resolution to full HD?

rahuljeyaraj, 2023-09-07:
Your project looks amazing. I would like to try it out. But I have a camera with full HD HDMI output. I could convert the HDMI input to axi stream, but looks like the resolution of input video is fixed in the project both in HW and SW to 640*480. Is it possible to increase the resolution to 1920 x 1080?

ztachip, 2023-09-07:
On SW, you need to change WEBCAM_WIDTH and WEBCAM_HEIGHT in SW/src/soc.cpp and SW/src/vision_ai.cpp

If you are still using VGA signals, you need to change the constants below in vga.vhd to reflect the VGA parameters for your new display resolution
constant FRAME_WIDTH:integer:=640;
constant FRAME_HEIGHT:integer:=480;
constant H_FP:integer:=48; -- H front porch width (pixels)
constant H_BP:integer:=16; -- H front porch width (pixels)
constant H_PW:integer:=96; -- H sync pulse width (pixels)
constant H_MAX:integer:=800; -- H total period (pixels)
constant V_FP:integer:=33; -- V front porch width (lines)
constant V_BP:integer:=10; -- V front porch width (lines)
constant V_PW:integer:=2; -- V sync pulse width (lines)
constant V_MAX:integer:=525; -- V total period (lines)

On camera, I don't know which camera you are using, you may need a new implementation for HW/examples/GHRD/camera.vhd since the one comes with ztachip is for OV7670 only

bittu-digital, 2024-07-30:
As I am changing the camera to higher resolution , what about the change in data rate ? Can you elaborate

ztachip, 2024-07-31:
In https://github.com/ztachip/ztachip/blob/master/HW/src/soc/soc_base.vhd
Change AXI-stream page size WRITE_PAGE_SIZE and READ_PAGE_SIZE = WIDTHxHEIGHTx3