ztachip/ztachip #9: Video Output got stuck

umesh-deepgrid, 2024-02-05:
[image]

After executing the program with the "continue" command, I observed that the output displayed on the monitor remains static for only one second. Subsequently, it becomes unresponsive and does not update, persisting in a frozen state even after pressing button 0.

ztachip, 2024-02-05:
Looks like your connection of webcam is not solid.

capture.jpg (view on web)<https://github.com/ztachip/ztachip/assets/108859093/add5a81e-ff05-44f3-a413-b35208c77f0a>

umesh-deepgrid, 2024-02-07:
I've examined all my connections, and they seem quite secure. However, despite this, the screen continues to freeze.

ztachip, 2024-02-07:
Try to run in test mode first to make sure firmware are ok
in SW/makefile change UNIT_TEST=true
Recompile everything
If your FPGA works, then you should have blinking led.
Try to also move/hold the pins to see if it is making any changes. Sometimes the pin connections are loose even if it looks solid.
I also apply some glue to keep the pins together.

explorer1100, 2024-02-08:
I adhered to the provided instructions, but unfortunately, there was no noticeable alteration. The only output I observed was a constant red and green LED illumination, with the green LED not exhibiting any blinking on the board. I have included a picture for your reference. Additionally, I have a question regarding a warning message I encounter when launching OpenOCD, which is attached here. I used 30cm Male to Female Jumper Wires to connect the camera to the board. I'm curious if the wire length might be a factor causing the issue and would appreciate clarification on this matter.
[image]
[image]

ztachip, 2024-02-08:
After loading program, confirm program is running using gnu debugger command
Also hit ctrl+c to see program still running or crashing. Or set breakpoint in main
Check that your tool chain is built correctl according to ztachip weɓsite

ztachip1.PNG (view on web)<https://github.com/ztachip/ztachip/assets/108859093/c77c7bb8-773b-4e91-9480-dbc79053c51e>
ztachip2.jpg (view on web)<https://github.com/ztachip/ztachip/assets/108859093/0ea69e1f-d073-4530-8dc9-e256a5f5f82b>

ztachip, 2024-02-09:
> I followed your suggestion and built Ztachip, but unfortunately, the problem persists. Following your instructions, I pressed Ctrl+C on the OpenOCD RISC-V command line after executing the continue command. However, the screen remains frozen even after exiting both the OpenOCD RISC-V and Ztachip command lines. Please find the attachments below for your reference.
> error1.PNG (view on web)<https://github.com/ztachip/ztachip/assets/108859093/d1efa412-ff25-49a5-93c9-537c5ce7cd58>
> error2.PNG (view on web)<https://github.com/ztachip/ztachip/assets/108859093/4170dfb0-c04d-4838-a929-5f8a486d3c72>

After loading program, instead of continue command, try step command to see how far it runs
Confirm that your tool chain is built with correct risc5 option

ztachip, 2024-02-12:
Please try latest ztachip release if it still does not work for you

explorer1100, 2024-02-13:
I just wanted to inform you that I'm working on it, and it will likely take a day or two. I'm starting fresh from scratch.

explorer1100, 2024-02-15:
Hi, as per your recommendation, I located the RISC-V version and compiler version. Here are the details:

```
explorer@ubuntu:~/ztachip/SW/src$ riscv32-unknown-elf-gcc --version
riscv32-unknown-elf-gcc () 13.2.0
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions. There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

explorer@ubuntu:~/ztachip/SW/src$ riscv32-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=riscv32-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/opt/riscv/libexec/gcc/riscv32-unknown-elf/13.2.0/lto-wrapper
Target: riscv32-unknown-elf
Configured with: /home/explorer/riscv-gnu-toolchain/gcc/configure --target=riscv32-unknown-elf --prefix=/opt/riscv --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion= --with-system-zlib --enable-tls --with-newlib --with-sysroot=/opt/riscv/riscv32-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././gcc --disable-multilib --with-abi=ilp32 --with-arch=rv32im --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 13.2.0 ()     
```

Following your advice, I utilized the step command, which resulted in a blank screen. Despite rebuilding ztachip, RISC-V, and openocd from the beginning, the freezing issue on the screen persists.

ztachip, 2024-02-16:
Try again with latest ztachip.
I have posted new version 2 days ago.
Before execute continue command, first check the memory content first to confirm SW loaded correctly with x command
It should look like below

(gdb) x 0
0x0 <crtStart>:   0x00002137
(gdb) x 4
0x4 <crtStart+4>: 0xb9410113
(gdb) x 8
0x8 <crtStart+8>: 0x000100e7
(gdb) x 12
0xc <crtStart+12>:      0x00000013
(gdb) x 16
0x10 <taskEntry>: 0xff010113
(gdb)

explorer1100, 2024-03-07:
Hello, I've managed to get it running smoothly now. I believe the problem was with my previous setup using Oracle VirtualBox, but now that I'm running it on a full Ubuntu desktop system, everything is working properly.

closed 2024-03-07 by explorer1100

ztachip, 2024-03-08:
Perfect
Please check in in a month's time, you will be able to run your applications from micropython.