ztachip/ztachip #11: Query Regarding Porting to following FPGA's

fabricchip, 2024-03-11:
Hello! Can we port ztachip to the following FPGAs with less than 100,000 LUTs:
1. Digilent Cmod A7-35T: Artix-7 with 35,000 LUTs
2. Cmod A7-15T: Artix-7 FPGA Module with 15,000 LUTs
3. CrossLink NX from Lattice Semiconductors (LIFCL-40, 39,000 LUTs)

Please let me know any steps to follow

ztachip, 2024-03-12:
Current version would fit a Xilinx Artix7-100 or Lattice ECP5-85
But I'm working on next version that would let you scale down to Artix7/35T. This is probably available by end of year.

fabricchip, 2024-03-12:
Hi

Awesome, thanks for the info!

The board perfectly works on arty a7 for now. We will report back on
lattice ecp5.

Falling is among the most damaging event elderly people may experience.
With the ever-growing aging population, there is an urgent need for the
development of fall detection systems. We belive your repo is the best use
case for this project.

We are also going to port few ai models starting with human activity
recognition check it's performance.

https://github.com/CMU-Perceptual-Computing-Lab/openpose

1.https://github.com/aminullah6264/Activity_Rec_ML-LSTM

2.https://github.com/aminullah6264/Activity_Rec_ML-LSTM

This we are doing to solve the problem elderly fall detection.

Performance Question:

I have few questions regarding is there in particular command I can type
like htop or top to see memory usage dynamically and performance metrics.

Regards
Fabric Team

fabricchip, 2024-03-24:
> Happy that you find the project useful.
> I also have an aunt who would have lived if she had the technologies you
> are working on.
> For computing performance, I could add something later to measure the ALU
> efficiency of pcores.
> For memory usage, ztachip is running on newlib which is bare-metal. But
> you can track heap usage in newlib.c although this only tracks peak usage.

Hi,

Yes thanks for the info. Elderly care is our aim in this standlone device.
Sorry to hear about your aunt. We belive this is the most neglected part of
our aim to solve this using standlone FPGA based edge device...we would
want to run this as low cost FPGA device eventually . For now we will stick
to arty a7.

 We have made significant progress in compiling the code.
I just wanted to understand to integrate.

https://www.kaggle.com/models/google/movenet/frameworks/tensorFlow2/variations/multipose-lightning

Since tensorflowlite is already supported on ztachip.

We belive the following steps are important for integration:

1.Model Compatibility Check: Verify that all operations used by the PoseNet
model are supported by the ztachip TensorFlow Lite implementation. If there
are unsupported operations, you might need to implement these at the
compiler level. Is this correct assumption?

2.Create New Application: In the SW/apps directory, create a new directory
specifically for your PoseNet application. This directory will contain all
the necessary code to run PoseNet.

3.Implement Application Logic: Use the existing TensorFlow Lite application
examples in SW/apps/nn as a reference. Adapt this logic to initialize and
run our PoseNet model, focusing on handling multi-person detection outputs.

3.Input and Output Processing: Implement logic for processing input (e.g.,
camera feed or static images) and for handling the output from PoseNet
(e.g., drawing skeletons on images or calculating pose scores).

4.Test and Validation:Run our application on the ztachip development kit
and Measure the performance (inference time, accuracy, etc.) of your
PoseNet model on the ztachip hardware.

Our Development Plan:

1.Load model Function: SW/apps/nn/model_loader.cpp
(Hypothetical.location)

2.PreprocessInput Function:This function would handle the conversion of
input images or video frames to the format expected by PoseNet, such as
resizing, normalization, and color conversion.
(preprocess.cpp)

3.RunInference Function: The core function where the model inference
happens. It involves feeding the processed input to the model and
retrieving the output.
Possible Location: SW/apps/nn/pose_net_inference.cpp (hypothetical path).

4.ParseOutput Function: After inference, this function interprets the
output tensor(s) from PoseNet to extract pose information, such as keypoint
locations and confidence scores.
Possible Location: SW/apps/nn/output_parser.cpp (hypothetical path)

5.DrawPoses Function: If you plan to visualize the results by drawing
keypoints and skeletons on images, you'll need a function for this purpose.
Possible Location: SW/apps/nn/visualize.cpp (hypothetical path)

6.Compiler Modifications (If Needed based on ztachip architect.feedback)
AddOperation Support: If PoseNet uses TensorFlow Lite operations not
supported by ztachip, you'll need to implement these in the compiler. This
involves modifying the lexer (Lexer.l), parser (Parser.y), and potentially
adding new instruction generation logic (gen.cpp).

Do we need to modify any existing compiler modifications for this
implementation?

Please do let me know if this process we are following is going good or any
modifications are necessary.

Regards
Team fabric

ztachip, 2024-03-24:
https://www.kaggle.com/models/google/movenet/frameworks/tensorFlow2/variations/multipose-lightning

I has a quick look at the graph for PoseNet, PoseNet looks like a typical CNN like Mobinet/SSD-Mobinet, ztachip should work fine but I am not very sure since I did not look very deep into it. Any new layers not supported yet are done in SW/apps/nn folder and not compiler. There is a file dedicated for each layer type, you need to implement a new layer type.

You dont really need a new folder for PoseNet since the GraphNode for TensorflowLite (TfliteNn) is generic, please reference to document below. I would put your new application under same folder like the example SW/src/vision_ai.cpp
https://github.com/ztachip/ztachip/raw/master/Documentation/visionai_programmer_guide.pdf

Same as question2, the way you use PoseNet is the same as the way MobiNet as shown in example SW/src/vision_ai.cpp, vision_ai.cpp is a good place to reference.

You can look at SW/src/vision_ai.cpp on how to do this, the example for SSD-mobinet draws rectangle around the detected objet

I normally use a test vector for verification, you can look at SW/src/test.cpp, there are example on how to unit test AI model with saved input and saved expected output.
To run unit tests, recompile the code with UNIT_TEST=yes in SW/makefile
For performance, next version there will be some function for it, but for now time benchmark how long it takes for graph completion would give you an idea.

vision_ai.cpp is a good example on how to start your app

There are example in vision_ai.cpp for image proprocessing

Same as above, vision_ai.cpp is a good reference for how to do it.

Same as above, vision_ai.cpp is a good reference for how to do it.

Same as above, vision_ai.cpp is a good reference for how to do it.

Compiler is for generic ztachip acceleration and not related to AI. There is no need to change there
Reference to answer of question1  above.

No

I think there is a good chance what you need to do: preprossing+posenet+display already supported by ztachip.
Seems like what you need to do is similar to what is done at vision_ai.cpp example.

ztachip, 2024-03-24:
In few weeks, there will be a new ztachip release with micropython support.
Then you also have many of the features available from micropython like networking, file system, web, ...
There is also good community support for micropython
It runs just as fast as native C/C++ since python is used primarily just to setup the graph, all processing is still native speed

https://www.kaggle.com/models/google/movenet/frameworks/tensorFlow2/variations/multipose-lightning

fabricchip, 2024-03-24:
> as shown in example , vision_ai.cpp is a good place to reference.

Wow that's quick and elaborate. So the main program is vision
SW/src/vision_ai.cpp.

Thanks for pointing out that and also no need to create a new folder
posenet , just a new layer file to be done under nn.

Rest is vision_ai.cpp

Thank you we will follow and update you.

Only one question: does ztachip parallel process work in "dynamic shape
input" since nvidia cuda cores don't work for dynamic shape input, in this
particular mutipose model.

{
https://www.kaggle.com/models/google/movenet/frameworks/tensorFlow2/variations/multipose-lightning
}

Handling Dynamic Input Shapes:

Dynamic Shape Support: Advanced AI models, including those for pose
estimation that handle varying numbers of detections, require support for
dynamic input shapes.
 The ztachip platform might have mechanisms in place to handle tensors with
dynamic shapes, adjusting memory allocation and operation execution as
needed.
Is this assumption correct?

aravindgp, 2024-04-01:
Wow you just released micropython code! "In few weeks, there will be a new ztachip release with micropython support.
Then you also have many of the features available from micropython like networking, file system, web.." Its good to learn this and I am sure our development will pickup pace on this micropython code you just released.

fabricchip, 2024-04-01:
Awesome you have released micropython as indicated earlier this makes the job very easy to port and try out new models. Especially our posenet model. We are actively working on it. I am closing the issue for now this thread now.

closed 2024-04-01 by fabricchip

ztachip, 2024-04-02:
The model https://www.kaggle.com/models/google/movenet/frameworks/tensorFlow2/variations/multipose-lightning
On the website it shows only FP16 version.
You need to do an extra quantization step to INT8 for ztachip to work however.
If you can get this converted, I can give it a try for you.

https://www.kaggle.com/models/google/movenet/frameworks/tensorFlow2/variations/multipose-lightning
}

fabricchip, 2024-04-02:
Sure, we will convert fp16 to int8 for multipose model. Its really worthy
the project.