ztachip/ztachip #14: About Benchmarking

FaizanAhmad626, 2024-04-14:
Hi Vuong,

It's me again from issues #6 and #5. I got back to this project recently and was revising everything.

I wanted to ask about measuring the performance of ztachip. My main goal is just to see/show how well ztachip is performing with AI workloads as compared to any other hardware architecture, may it be standalone Vexriscv or Nvidia GPU.

As you mentioned in the readme file:

> Acceleration provided by ztachip can be up to 20-50x compared with a non-accelerated RISCV implementation on many vision/AI tasks. ztachip performs also better when compared with a RISCV that is equipped with vector extension.

And in the overview file:

> Using the popular MobileNet-SSD AI model as a reference point, ztachip achieves a performance of 10fps at 20GOPS of hardware computing resources. Compared with Nvidia Jetson Nano, it has a performance of 40fps but with a computing hardware resource at 500GOPS. Therefore ztachip has a 6x better computing resource utilization than Nvidia in this case, resulting in much lower power consumption.

I simply want to prove these results again, if possible. Or in any way generate results that are comparable with other hardware architectures and outrun them. It would be great to compare it with RISC-V processor with vector extension but it's also great with Nvidia Jetson Nano. I haven't performed benchmarking of any hardware before so can you please guide me in this?

PS: I am still using that older version of the project, which is working fine, and not this newer one with micropython support. Although I tried running the latest one but it wasn't working, maybe because I made some mistake while porting.

Regards,
Faizan

ztachip, 2024-04-14:
If you use latest version, it is not just easier to use but also a lot faster
Also there is a benchmark number printed on the screen.
https://www.youtube.com/watch?v=nLGmmw7-PYs&t=4s

Examples python scripts used in demo above are
https://github.com/ztachip/ztachip/tree/master/micropython/examples

In the example, it is easy to get benchmark number
       zta.CanvasDrawText('{} msec'.format(zta.GetElapsedTimeMsec()),0,0)

ztachip, 2024-06-22:
You resolve the issue of JTAG connection before
You mind sharing how you fixed it. I have someone with similar problem
Thx

FaizanAhmad626, 2024-06-22:
Yes. As far as I can remember I changed this line [ftdi_channel 0](https://github.com/ztachip/ztachip/blob/master/tools/openocd/usb_connect.cfg#L22) to **ftdi_channel 1**.

In some cases you may have to change the device name too, [ftdi_device_desc "Digilent USB Device"](https://github.com/ztachip/ztachip/blob/master/tools/openocd/usb_connect.cfg#L20) to the name shown on your VM if using one. In my case I changed it to **ftdi_device_desc "Digilent Adept USB Device"**

I'll share the [usb_connect.cfg](https://github.com/ztachip/ztachip/blob/master/tools/openocd/usb_connect.cfg) file I used.
[usb_connect.txt](https://github.com/user-attachments/files/15938048/usb_connect.txt)
It is in the form of text as GitHub doesn't support config files.

Let me know if need further help.
Regards,
Faizan.