ztachip/ztachip #16: roadmap of the feature and throughput

jinsong-mao, 2024-06-19:
Hi 

what's the roadmap of the features and any site about the throughput data?

Thanks

ztachip, 2024-06-22:
Roadmap:
- Convert to spinalHDL for easy customization (different data types) and Verilog generation (good for opensource ASIC toolchain)
- Matrix data type support (instead of just vector currently) for faster throughput (this would be equivalent to Nvidia's Tensor core)
- Data compression support
- Upgrade compiler backend, perhaps to LLVM

jinsong-mao, 2024-06-24:
Thanks @ztachip 

It's much clear to me,  maybe a table or a list of items which includes all the features supported or to be supported in the home page would be much better. 

and if there is one table contains the current supported networks with measured throughput on specific FPGA device would be help a lot on get the first impression of performance. 

Thanks

chili-chips-ba, 2024-07-05:
We are curious to learn the full story on the reasoning that made you select SpinalHDL for the future dev work on this core.

ztachip, 2024-07-05:
We like to move away from VHDL since opensource toolchain supports just basic Verilog mostly.
SpinalHDL can generate basic verilog while I still can get all the advanced features normally found only in SystemVerilog.
Also it is easier for users to customize ztachip without having to change the base code.

chili-chips-ba, 2024-07-06:
We would seriously recommend to reconsider. 

While it is true that Yosys does not seem to have much interest in System Verilog, and that Synlig plugin has serious problems, there are other venues to pursue, such as `SV2V`. Also see [this](https://github.com/chili-chips-ba/openCologne/issues/3#issuecomment-2169002938).

Essentially, if RTL developers don't keep asking for, and working towards finding a way to get proper System Verilog into opensource, it won't happen. And, it is much harder and unfair to ask new grad engineers to invest time and effort in learning SpinalHDL, given that they cannot find industry jobs with it. [Amaranth](https://github.com/amaranth-lang/amaranth) and [Chisel](https://github.com/chipsalliance/chisel) would in that sense be a better choice. Yet, SystemVerilog is still the best of all for today's chip design industry.

ztachip, 2024-07-06:
SV2V looks very interesting.
This will also solve the problems for open-source toolchain support.
I will check this out.
Many thanks for your recommendations.

chili-chips-ba, 2024-07-06:
@zachjs, @phsauter, take note of this potential new user of your SV preprocessors, and for a processor design that's equally involved and elaborate as PULP and [lowRISC](https://github.com/lowRISC/ibex/tree/d019dccb4b6fb1a580cefabff0391a00bb123ffb/syn) processors 😉