ztachip/ztachip #18: training model in python and loading on to ztachip

RickSancheeez, 2024-07-05:
How can I train a model in Python and load it in Ztachip as I am not "import zta" in Micropython? Can you please explain the complete flow?

ztachip, 2024-07-05:
Not sure I understand your question, you mean you don't want to run in micropython?
And your model, is it a custom model?

RickSancheeez, 2024-07-05:
I cannot understand the exact flow of the software part, and I didn't find any zta module when I tried to import it. 
Can you explain the flow of it so that I can run my custom models on this.

ztachip, 2024-07-05:
Sure, I will update the doc and add some python functions to handle custom model.

aashrith-deepgrid, 2025-04-25:
Hello. I also would like to run a custom python model, can you please point me to the updated docs.

ztachip, 2025-04-26:
To run TensorFlow with ztachip, follow instructions from the sections below on github page
   - Build ztachip as micropython port
   - Upload procedure for micropython SW image option

You can run any custom Tensorflow model providing that your model uses the layer types found in mobinet and SSD-mobinet. These 2 models cover the essential layers of CNN.

There are 2 ways to run ztachip. Either via C-API or MicroPython.

Running TensorFlow model from a C-API is described in link below https://github.com/ztachip/ztachip/raw/master/Documentation/visionai_programmer_guide.pdf

I only have micropython hook for mobinet and SSD-mobinet
To add micropython hook to your custom model, you can follow how I did for other models.
Modify the file https://github.com/ztachip/ztachip/blob/master/micropython/ztachip_port/zta.c to add new custom python hook for your custom model