ztachip/ztachip #19: Verilog version

bittu-digital, 2024-07-05:
where can I find the verilog versions of the ztachip as Iam not familiar with the VHDL

ztachip, 2024-07-05:
Im working on a spinalHDL port of ztachip.
With spinalHDL, then you can then generate to Verilog
Probably available by year-end.

chili-chips-ba, 2024-07-06:
@ztachip, have you considered the much more mainstream SystemVerilog instead of niche SpinalHDL?!

With SV Interfaces and Structs, you can have about the same expressiveness as SpinalHDL, i.e higher than current VHDL, but much wider reach and audience. 

When your sources are in SpinalHDL, Amaranth, PipelineC, or another sidestream HDL, the auto-generated mainstream Verilog is rather unreadable...

ztachip, 2024-07-06:
Im really interested in getting ztachip ported to opensource ASIC tool chain however.
You think opensource ASIC toolchain will support SystemVerilog?
Otherwise the cost for ASIC tool chain will be prohibitively expensive for opensource project.
Actually with spinalHDL, you dont really need to look at the generated Verilog, like when you program in "C", you dont need to look at the generated assembly. At least this is my understanding.

chili-chips-ba, 2024-07-06:
... just think about it: `ASIC is way more physical than the FPGA`, isn't it?! 

Now, imagine having to understand and solve DFT, STA, Floorplanning and other PNR issues at the netlist level when your RTL sources are already looking like an obfuscated netlist. All that for a complex design which is pushing the Fmax and LP envelope. If you are thinking of a zero-effort, hands-off, push-button back-end flow, think twice.

Given that the entire opensource toolchain is based on Verilog, transition from SystemVerilog to Verilog is more natural and simpler than a jump from anything else to Verilog.

Also check CHIPS ALLIANCE [Verible](https://github.com/chipsalliance/verible) Linter (@hzeller is in the best position to further elaborate) along with their 3 RISC-V cores, all written in SV:
- https://github.com/chipsalliance/Cores-VeeR-EL2#readme
- https://github.com/chipsalliance/Cores-VeeR-EH2#readme
- https://github.com/chipsalliance/Cores-VeeR-EH1#readme

[lowRISC](https://github.com/lowRISC/ibex/tree/d019dccb4b6fb1a580cefabff0391a00bb123ffb/syn) core is also written in a good, rich command of SystemVerilog, yet processed with opensource tools from over 4 years ago.

As for the tapeout, we recommend the [IHP](https://ihp-open-ip.readthedocs.io/en/latest/) free shuttle program...

ztachip, 2024-07-06:
I was hoping more for what openRoad promised with 24 hour turn around chip production , for small design but less than optimized chip design.
https://github.com/The-OpenROAD-Project
I think there is a need for low-end ASIC with easy entry point. So we will be less reliant on "black box" chips that keep getting obsolete and running out of stock.
But I agree that it is better to have support both for people who are looking for performance and for people who just need something fast and cheap.
Your recommendation to start with SystemVerilog and convert to Verilog for low-end design is a very good idea and I'm looking into this option since this will cover both camps.
Thx

Also check CHIPS ALLIANCE Verible<https://github.com/chipsalliance/verible> Linter ***@***.***<https://github.com/hzeller> is in the best position to further elaborate) along with their 3 RISC-V cores, all written in SV:

  *   https://github.com/chipsalliance/Cores-VeeR-EL2#readme
  *   https://github.com/chipsalliance/Cores-VeeR-EH2#readme
  *   https://github.com/chipsalliance/Cores-VeeR-EH1#readme

ztachip, 2024-08-18:
I managed to convert from VHDL to verilog using GHDL. 
There are some features not supported by GHDL but I found work-around for it.
Latest update will have what you need for an all-verilog RTL code base
Refer to link below for instructions on how to convert to Verilog. Refer to section [Support for opensource toolchain]
https://github.com/ztachip/ztachip/blob/master/Documentation/Vivado.md
This is not ideal, but eventually RTL code base will be rewritten to system-verilog.

kevinli-hw, 2025-04-05:
Thanks for your contribution. I followed the guide at https://github.com/ztachip/ztachip/blob/master/Documentation/Vivado.md, but I encountered numerous errors during the conversion. Here is the log:
`
../../HW/src/soc/soc_base.vhd:30:10: unit "numeric_std" not found in library "ieee"
../../HW/src/config.vhd:9:10: unit "numeric_std" not found in library "ieee"
../../HW/src/ztachip_pkg.vhd:28:10: unit "numeric_std" not found in library "ieee"
../../HW/src/ztachip_pkg.vhd:39:18: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:41:41: subtype "tid_t" is not visible here
../../HW/src/ztachip_pkg.vhd:269:61: subtype "tid_t" is not visible here
../../HW/src/ztachip_pkg.vhd:328:32: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:330:32: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:332:56: subtype "iregister_t" is not visible here
../../HW/src/ztachip_pkg.vhd:340:30: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:342:54: subtype "iregister_auto_t" is not visible here
../../HW/src/ztachip_pkg.vhd:348:35: subtype "tid_t" is not visible here
../../HW/src/ztachip_pkg.vhd:532:19: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:534:20: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:536:41: subtype "page2_t" is not visible here
../../HW/src/ztachip_pkg.vhd:536:50: no declaration for "to_unsigned"
../../HW/src/ztachip_pkg.vhd:538:41: subtype "page2_t" is not visible here
../../HW/src/ztachip_pkg.vhd:538:50: no declaration for "to_unsigned"
../../HW/src/ztachip_pkg.vhd:540:41: subtype "page2_t" is not visible here
../../HW/src/ztachip_pkg.vhd:540:50: no declaration for "to_unsigned"
../../HW/src/ztachip_pkg.vhd:542:41: subtype "page2_t" is not visible here
../../HW/src/ztachip_pkg.vhd:542:50: no declaration for "to_unsigned"
../../HW/src/ztachip_pkg.vhd:548:18: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:556:18: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:562:20: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:568:30: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:570:35: subtype "avalon_bus_page_t" is not visible here
../../HW/src/ztachip_pkg.vhd:570:54: no declaration for "to_unsigned"
../../HW/src/ztachip_pkg.vhd:572:34: subtype "avalon_bus_page_t" is not visible here
../../HW/src/ztachip_pkg.vhd:572:53: no declaration for "to_unsigned"
../../HW/src/ztachip_pkg.vhd:574:34: subtype "avalon_bus_page_t" is not visible here
../../HW/src/ztachip_pkg.vhd:574:53: no declaration for "to_unsigned"
../../HW/src/ztachip_pkg.vhd:576:34: subtype "avalon_bus_page_t" is not visible here
../../HW/src/ztachip_pkg.vhd:576:53: no declaration for "to_unsigned"
../../HW/src/ztachip_pkg.vhd:588:49: subtype "tid_t" is not visible here
../../HW/src/ztachip_pkg.vhd:594:50: subtype "page_t" is not visible here
../../HW/src/ztachip_pkg.vhd:606:59: subtype "avalon_bus_page_t" is not visible here
../../HW/src/ztachip_pkg.vhd:612:35: subtype "cid_t" is not visible here
../../HW/src/ztachip_pkg.vhd:616:42: subtype "cid_t" is not visible here
../../HW/src/ztachip_pkg.vhd:634:21: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:636:46: subtype "vector_t" is not visible here
../../HW/src/ztachip_pkg.vhd:666:23: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:668:24: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:670:38: subtype "burstlen_t" is not visible here
../../HW/src/ztachip_pkg.vhd:672:48: subtype "burstlen_t" is not visible here
../../HW/src/ztachip_pkg.vhd:700:24: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:724:27: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:726:38: subtype "dp_data_type_t" is not visible here
../../HW/src/ztachip_pkg.vhd:728:38: subtype "dp_data_type_t" is not visible here
../../HW/src/ztachip_pkg.vhd:730:52: subtype "dp_data_type_t" is not visible here
../../HW/src/ztachip_pkg.vhd:751:24: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:753:49: subtype "stream_id_t" is not visible here
../../HW/src/ztachip_pkg.vhd:779:14: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:780:20: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:781:18: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:782:18: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:783:14: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:784:20: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:785:18: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:786:18: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:787:14: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:788:20: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:789:18: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:790:18: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:791:14: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:792:20: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:793:18: no declaration for "unsigned"
../../HW/src/ztachip_pkg.vhd:794:18: no declaration for "unsigned"
[... 64 lines cut ...]
../../HW/src/soc/soc_base.vhd:712:43: conversion function or type does not match
../../HW/src/soc/axi/axi_merge.vhd:27:10: unit "numeric_std" not found in library "ieee"
../../HW/src/soc/axi/axi_apb_bridge.vhd:27:10: unit "numeric_std" not found in library "ieee"
../../HW/src/soc/axi/axi_stream_write.vhd:27:10: unit "numeric_std" not found in library "ieee"
[... cut ...]
../../HW/src/soc/peripherals/gpio.vhd:21:10: unit "numeric_std" not found in library "ieee"
../../HW/src/soc/peripherals/uart.vhd:21:10: unit "numeric_std" not found in library "ieee"
../../HW/src/soc/peripherals/vga.vhd:21:10: unit "numeric_std" not found in library "ieee"
../../HW/src/soc/peripherals/camera.vhd:21:10: unit "numeric_std" not found in library "ieee"
`
I'm not familiar with the tool GHDL. Does this mean that I missed some configuration when using `./convert.sh`, or I should download some library for the script?

ztachip, 2025-04-06:
There is probably a problem with your version of GHDL
I have included my version of GHDL binary in ztachip repo. It should work.

kevinli-hw, 2025-04-06:
Thanks for the reply. Just as you said, the GHDL version is the reason. I also checked the issue on the GHDL repo; [issue](https://github.com/ghdl/ghdl/issues/1686) explains why lower versions (<1.0) could cause the error. I upgraded my GHDL (0.37) to 4.0, and now the `convert.sh` works fine.

The next question is about the Verilog project. I now create the project with soc.v., the synthesis part is fine, but the implementation fails, saying `[Route 35-2] Design is not legally routed. There are 81084 node overlaps.` The VHDL project doesn't have the error, so I guess this is still because I miss some configurations. It would be great if you could give some suggestions.

ztachip, 2025-04-07:
Did you build with Vivado?
Also try with GHDL that comes with ztachip repo (under ztachip/tools/ghdl/ghdl)

kevinli-hw, 2025-04-11:
Sorry for the late reply.

> Did you build with Vivado?

Yes, I'm using Vivado 2024.2.

> Also try with GHDL that comes with ztachip repo (under ztachip/tools/ghdl/ghdl)

I tried the GHDL from the repo, which is 5.0.0. Following the manual, I used the tcl script to create the project with the generated soc.v. Though I didn't receive the previous message, the implementation still failed due to some critical warnings, something related to `debug_hub` in the constraints.

ztachip, 2025-04-12:
Remove the lines below in HW/examples/GHRD/main.xdc. They are for signal tap debugging which you don't need.

set_property C_CLK_INPUT_FREQ_HZ 300000000 [get_debug_cores dbg_hub]
set_property C_ENABLE_CLK_DIVIDER false [get_debug_cores dbg_hub]
set_property C_USER_SCAN_CHAIN 1 [get_debug_cores dbg_hub]
connect_debug_port dbg_hub/clk [get_nets clk]

kevinli-hw, 2025-04-14:
Thanks for your patience. Now the implementation succeeds after removing these lines. The design failed to meet the timing requirements, but I see your demo runs correctly. Is it necessary to fix all the violations if I want to deploy my own model?

ztachip, 2025-04-14:
The timing violation is usually for worst case under highest temperature. If you are not at that operating range, the current timing violation is within tolerable range.
But if you want to remove all warning to work at any conditions, you may need to reduce the clock from 125M to 100M. Clock is defined in the script when you create the project.