ztachip/ztachip #20: APB interface?

chili-chips-ba, 2024-07-06:
Hi @ztachip, have you, for the sake of performance and simplicity, considered replacing 2-cycle APB with a generic VALID/READY interface?! 

That latter can perform transaction in a single-cycle, and is even lighter on resources than APB.

ztachip, 2024-07-06:
APB is not efficient but can be useful for the case you need to be interop with other IPs which probably complied to AXI specs.

chili-chips-ba, 2024-07-06:
What exactly is the other IP that ztachip design is pulling in, and that comes with APB?

ztachip, 2024-07-06:
Not currently.
Now the only external IP are RISCV (from VexRiscv) and memory-controller (from Xilinx)
But I may add support for Wishbone later. So I would need something different than APB for peripheral bus. I have not looked into Wishbone yet, but are your VALID/READY signals similar to Wishbone peripheral bus?

chili-chips-ba, 2024-07-08:
What we are recommending is a simple, lightweight, custom interface that you can adapt to the needs of design, and so strike the perfect balance between cost and performance. Here is interface that we typically use for our SOCs. 
```
interface soc_if (
   input  logic arst_n, // asynchronous, active-low reset  
   input  logic clk     // bus clock
);

   import soc_pkg::*;
   
   logic        vld;    // 1 for access REQuest     -\ transacion is performed
   logic        rdy;    // 1 for access ACKnowledge -/ when both vld&rdy are 1

   soc_addr_t   addr; 
   soc_we_t     we;     // access is WRITE for (vld & |we). READ for (vld & ~|we)
   soc_data_t   wdat;
   soc_data_t   rdat;

  //---------------------------------------- 
  // master/CPU side
  //---------------------------------------- 
   modport MST (
     output vld,
            addr,
            we, wdat,

     input  arst_n, clk,
            rdy,
            rdat
   );

  //---------------------------------------- 
  // Slave/Peripheral side
  //---------------------------------------- 
   modport SLV (
     input  arst_n, clk,
            vld,
            addr,
            we, wdat,

     output rdy,
            rdat
   );

endinterface: soc_if
```
As you can see, it has per-byte Write_Enable, which is what RISC-V needs. DDR Memory Controllers typically also need it...

ztachip, 2024-07-13:
ztachip has 3 external interfaces to do

  *
connect to a DDR memory controller (AXI-MM)
  *   memory-mapped interface to peripherals (AXI-APB)
  *
Streaming interface to stream data to/from camera/display (AXI-Stream)

I will try to give more choices to these interfaces in future releases like Wishbone and Native (Native can be something simpler like you said)

chili-chips-ba, 2024-07-13:
What is the internal interface between CPU (VexRiscV) and PCores AI accelerator?

The question is for both:
- low-level physical handshake signals
- and logical level communication over it

ztachip, 2024-07-14:
RISCV D-BUS (AXI based) is used to communicate between RISCV and AI accelerator.

chili-chips-ba, 2024-07-14:
Where can we find more info about that "AXI-based" interface?!
 1) Is it AXI-Lite, or AXI4-MM, or something else?
 2) How about the format and contents of data transfers?
> - Is it a bunch of memory-mapped registers
> - or some kind of messaging protocol
> - or something else, perhaps even a hybrid / cross between multiple things?

ztachip, 2024-07-14:
RISCV D-BUS is AXI-MM/Full. But when RISCV communicates with ztachip, it is in LITE mode (non-burst)
ztachip interface to RISCV is a set of memory-mapped registers for RISCV to emit tensor instructions to ztachip core. So it should work with any RISCV implementation.

You can find out more information about ztachip HW design from
https://github.com/ztachip/ztachip/blob/master/Documentation/HardwareDesign.md

For overview and SW
https://github.com/ztachip/ztachip/blob/master/Documentation/Overview.md
https://github.com/ztachip/ztachip/raw/master/Documentation/ztachip_programmer_guide.pdf
https://github.com/ztachip/ztachip/raw/master/Documentation/visionai_programmer_guide.pdf
https://github.com/ztachip/ztachip/blob/master/micropython/MicropythonUserGuide.md