ztachip/ztachip #21: RTL Linting as the first QA flow step!

chili-chips-ba, 2024-07-19:
While we don't know about good open-source linters for VHDL and SpinalHDL, **(System)Verilog** has got at least three: **Verilator**, **Verible**, **Slang**. Check [this](https://github.com/chili-chips-ba/openCologne/issues/16#issuecomment-2229927560) for some background. 

Once ZTACHIP codebase is moved to SV, we strongly recommend introducing Linting into design and QA flow. Verible can also help with code formatting. 

The next steps could be to introduce QA gatekeepers for RTL commits:
- 1st level: Lint pass
- 2nd level: Regression pass

chili-chips-ba, 2024-07-19:
Heads up, we may also get a [CDC linter](https://github.com/YosysHQ/yosys/issues/4493#issuecomment-2238977629) for Verilog

ztachip, 2024-07-20:
This is very helpful info. Thx