ztachip/ztachip #22: More peripherals

sai-deepgrid, 2024-07-25:


ztachip, 2024-07-25:
Peripherals are added at top component. There are comments in the file
https://github.com/ztachip/ztachip/blob/master/HW/examples/GHRD/main.vhd
For SW drivers for these peripherals, you can reference some example in
https://github.com/ztachip/ztachip/blob/master/SW/src/soc.cpp

digee-bytes, 2024-07-29:
My query is, "How can I connect more cameras to the system or use a camera with a better-quality image sensor?"

chili-chips-ba, 2024-07-29:
See [this](https://github.com/chili-chips-ba/openeye-CamSI) for the camera CSI interface part, which is half of the solution. This other half is interface to ZTACHIP itself...

digee-bytes, 2024-07-30:
@chili-chips-ba Thank you, @ztachip, but can you elaborate more on using "like a better camera for processing and changes needed in the code in hardware and software flow" for this?

ztachip, 2024-07-31:
Please reference
https://github.com/ztachip/ztachip/issues/7