ztachip/ztachip #23: About pcore number and documentation

KalamataTech, 2024-07-28:
Hello, thank you for your hard work and open source code
I see that the readme description of tensor has 28 Pcores, but I found in the documentation and code that 2 cells are instantiated, each cell has 4 Pcores, and there are only 8 Pcores in total, am I wrong?
And could you please update all the documents? Thank you for your hard work.

ztachip, 2024-07-28:
ztachip design supports a large number of pcores.
It is currently limited to 8 due to FPGA resources available of Artix-7

KalamataTech, 2024-07-29:
Thank you for your response. 
Do you have any plans for updating the document? Thank you once again.

ztachip, 2024-07-31:
There are several documents available at https://github.com/ztachip/ztachip under Documentation section.

FaizanAhmad626, 2026-01-14:
I had a similar question about 28 pcores but now it's answered. Thanks.
I have a Kintex-7 fpga. Maybe it'll implement with 28 pcores on this fpga. Can you please guide how?
I figured I just have to change the value of cid_gen_max_c somehow.