ztachip/ztachip #24: Micropython

bittu-digital, 2024-07-30:
git clone https://github.com/micropython/micropython.git
cd micropython/ports
cp -avr <ztachip installation folder>/micropython/ztachip_port .
cd ztachip_port
export PATH=/opt/riscv/bin:$PATH
export ZTACHIP=<ztachip installation folder>
make clean
make

these are the commands which i followed to build ztachip as micropython port. But I got the error when i ran the last step i.e, make.
I donot know how to resolve this error. It will be very helpful on resolving this 
[image]

ztachip, 2024-07-31:
Set below
export ZTACHIP=<Directory where ztachip is installed>
Note that ztachip must be built first before micropython build

bittu-digital, 2024-07-31:
I have done the build process , standalone image is working fine.
While going with micro-python , 
export ZTACHIP=<given the path , where zta-chip is built>.
But the error i got is same again after "make"

Even while opening minicom using " sudo minicom -w -D /dev/ttyUSB1" i could not open any prompt to paste the example codes.

ztachip, 2024-07-31:
You seem to run a different version of RISCV compiler.
Try to rebuild RISCV tool chain with instruction under section "Download and build RISCV tool chain"

ztachip, 2024-07-31:
FPGA must be reset after you connect the serial port since it disrupts the JTAG connection of RISCV core.

bittu-digital, 2024-08-02:
I downloaded the same RISCV tool chain which worked for the standalone image, and Also I did reset the FPGA. I got the same error.
Can you provide me the detailed process for micro-python, may be I am missing something.

ztachip, 2024-08-03:
Tensorflow support is implemented under...
https://github.com/ztachip/ztachip/blob/master/SW/apps/nn/nn.cpp
The layer currenty supported are shown in function NeuralNet::CreateLayer which are the layers required for Mobitnet and SSD-Mobinet
If face detection uses these same layer types than it should work already. You can take a look at example in SW/src/vision_ai.cpp on how to load and run TensorFlow model.
Otherwise, you need to port the missing layers from TensorFlowLite to ztachip if the new layers are not taking much computing, otherwise, the new layers need an accelerated version using ztachip programming language.

ztachip, 2024-08-03:
If you still have trouble with micropython build, please try latest version and copy over the new ztachip_port folder

closed 2024-10-28 by bittu-digital