ztachip/ztachip #27: FPGA Test

kevinli-hw, 2025-04-26:
I'd like to run ztachip on my FPGA board, but unfortunately, I don't have any camera modules at the moment. I'm wondering if it is possible to remove the VGA & camera module, just load the program & test image, and receive the result (like SoftMax result vector) from USB.

ztachip, 2025-04-26:
Yes. It is possible
You can take an example at SW/src/test/test.cpp
This is a test program to verify ztachip using just test vectors.

kevinli-hw, 2025-04-30:
Thanks for the reply. Currently, before the FPGA test, I'd like to run ztachip in simulation to see how those vector operations are executed. I'm using VCS and following the instructions to include the needed sources. However, there are some warnings during the analysis part.

_1) NULL RANGE_
```
Warning-[AGGRNULLRANGE] Null range
../../src/v1_25_04_2025/src/top/ddr_rx.vhd, 598
DDR_RX_BEHAVIOR

  ...r_bus_width_c-1 downto dp_full_addr_width_c+data_byte_width_depth_c) :...
                                                                          ^
  Range 31 downto 32 is a null range.

"../../src/v1_25_04_2025/src/top/ddr_rx.vhd": errors: 0; warnings: 1.
Parsing design file '../../src/v1_25_04_2025/src/top/ddr_tx.vhd'

Warning-[AGGRNULLRANGE] Null range
../../src/v1_25_04_2025/src/top/ddr_tx.vhd, 269
DDR_TX_BEHAVIOR

  ...r_bus_width_c-1 downto dp_full_addr_width_c+data_byte_width_depth_c) :...
                                                                          ^
  Range 31 downto 32 is a null range.

"../../src/v1_25_04_2025/src/top/ddr_tx.vhd": errors: 0; warnings: 1.
```
It seems like the assignment for signal `ddr_addr_v` in `ddr_rx.vhd & ddr_tx.vhd` has some problems.

_2) LIBRARY ALTERA_MF_
```
Warning-[VHDL-NOLIBRARYDIR] Unable to use directory
./sim_src/simulation/SPRAM.vhd, 28

  LIBRARY altera_mf;
          ^
  Unable to use the logical name 'ALTERA_MF' that is mapped to directory
  'ALTERA_MF' because the directory does not exist.

"./sim_src/simulation/SPRAM.vhd": errors: 0; warnings: 1.
Parsing design file './sim_src/simulation/SPRAM_BE.vhd'

Warning-[VHDL-NOLIBRARYDIR] Unable to use directory
./sim_src/simulation/SPRAM_BE.vhd, 28

  LIBRARY altera_mf;
          ^
  Unable to use the logical name 'ALTERA_MF' that is mapped to directory
  'ALTERA_MF' because the directory does not exist.

"./sim_src/simulation/SPRAM_BE.vhd": errors: 0; warnings: 1.

```
I didn't find this library in the repo files, so I installed Quartus Prime to get this simulation library. I'm wondering if you could share more details on the generation of the sim_lib for VCS. If you're using any other simulation tools, please let me know. Also, as you provide both Xilinx and Altera wrappers, I'm wondering if those Xilinx wrappers are able to simulate. 

_3) PCORE DESIGN_
This is not a simulation issue, just to discuss the arch. Currently, the design treats AI operators as a series of vector operations, and handles them with 2 hardware threads. I'm wondering why the inner arch of pcore is a series of alus, not like systolic arrays(SA) used in common AI accelerators. My next step after simulation and FPGA test is to optimize ztachip and add support for transformer operators. I would be very grateful if you could offer some advice.

ztachip, 2025-04-30:
I will fix the warning below on next release. But it should be harmless however.
I am using ModelSim for simulation. Altera provides a free version with their installation.
You can just remove the reference to library ALTERA_MF, simulation package is pure VHDL code without any external dependencies
PCORE is vector now, but it is my goal to make it a matrix later. When implemented, you don't have one large systolic like Google's Edge TPU but rather 64 smaller systolic (8 systolic per core) of dimension 8x8. I started out with vector since it requires only 2 level of memory hierarchy (DDR and TCM). But with matrix, to keep up with higher memory bandwidth, it needs a third level of memory hierarchy which could be register type of memory within PCORE/ALU. So it will take more resources.
I am actively working on supporting transformers. I should have it ready be end of the year. The first model for reference will be smolVLM from HuggingFace. The main function to add is support for mixed precision (between float and integer)

kevinli-hw, 2025-05-01:
Thanks for the detailed reply! I'm really looking forward to the next version.

kevinli-hw, 2025-05-02:
There is one more bug I met during the simulation. I replace the Altera memory wrappers with Xilinx ones, and copy the needed files from the Vivado IP directory. However, the log said it required the shared library。
```
./simv: error while loading shared libraries: _429053_archive_1.so: cannot open shared object file: No such file or directory
```
I'm not sure if this is because I missed some settings of VCS, or I missed some library files to run the simulation.

kevinli-hw, 2025-05-06:
Thanks for the reply. Now I have changed the wrapper files and commented out the "Altera" lines. However, still facing the similar error.
```
./simv: error while loading shared libraries: libvhdlobjs.so: cannot open shared object file: No such file or directory
```
It seems like I missed some environment settings. I'll try to use ModelSim for simulation.

ztachip, 2025-05-07:
You can also try verilator. Very popular one. And it is also very fast. But it supports only Verilog.
Follow instructions on ztachip github page to convert the project ( VHDL based) to verilog and it should work with Verilator

kevinli-hw, 2025-05-07:
Thanks for the advice! Actually, I did use ghdl to convert the project to soc.v. However, after conversion, some signals' names become incomprehensible. So, I turned to the original source files. But I'll try this method if I cannot fix my bug.

ztachip, 2025-05-07:
I use this version of ModelSim from Intel. Its free
ModelSim 10.5b available with the installation of QuartusPrime 17.0
With later version, it seems Intel put some restrictions so its run slower

kevinli-hw, 2025-05-07:
I see. Thanks for sharing! I'll try it.

ztachip, 2025-05-08:
> I was trying to use the same version of Quartus Prime, but when I installed it, I found that it only supports Red Hat 5/6/7, while the server I'm using is Red Hat 8.9. Then I tried later versions (23/24), however, they don't include ModelSim in the installation. So as a temporary solution, I'm using ModelSim 20.01 to run the simulation. When Modelsim loaded the design, it threw the following error:
>
> # ** Error: (vsim-3033) Instantiation of 'BSCANE2' failed. The design unit was not found.
> #    Time: 0 ps  Iteration: 0  Instance: /sim_main/soc_base_inst/cpu_inst File: ../../src/v1_25_04_2025/xilinx_jtag/riscv.v Line: 225
> #         Searched libraries:
> #             /mnt/storage/local/users/yike/bert_acc/HW/sim/v1_25_04_2025/work
>
> I didn't find any files in the ztachip repo that have a definition of 'BSCANE2', so I'm wondering if I still missed some libraries for simulation. Meanwhile, I'll also try to use Quartus Prime 17.0 on Windows, see if everything works fine.

You should use the riscv generated for simulation
Use ztachip/HW/riscv/sim/riscv.vhd instead
This riscv is generated by VexRiscv project and defined in ztachip/HW/riscv/sim/riscv.scala

kevinli-hw, 2025-05-08:
Sorry, I didn't check the instructions carefully. Now it works fine. Thanks for the reply.