ztachip/ztachip #28: Video Zoom-In/Out (scaling) example?

chili-chips-ba, 2025-06-04:
Is there a good example of video scaling with ztachip, ideally for up to 4K @60Hz?!

This is not an issue, but a mere question. Hope it is of interest for the wider community...

ztachip, 2025-06-04:
I have example for scale down. Reference
https://github.com/ztachip/ztachip/raw/master/Documentation/visionai_programmer_guide.pdf

But it would not be hard to do scale up using ztachip programming API
https://github.com/ztachip/ztachip/raw/master/Documentation/ztachip_programmer_guide.pdf

chili-chips-ba, 2025-06-05:
How is the performance within standard Artix-7 target?

Can it do 4K, 60FPS, RGB888? If it can, is there horsepower left for the 2nd channel of the same, such as for stereo cameras?!

chili-chips-ba, 2025-06-05:
Are we saying that on-chip RAM is insufficient in size for this task, or also hard to tap into by the ztachip design?

ztachip, 2025-06-05:
I meant DDR memory bandwidth.
ztachip uses on-chip RAM for temporary buffer only but the bulk of data is in external DDR
A 4K video frame is taking 20MByte, too much for on-chip memory
I am working on next version ztachip that extends ztachip AXI bus to DDR controller to 128-bit. I think this will work better for 4K
Probably ready by end of the year.

chili-chips-ba, 2025-06-06:
... yes, it is understood that the off-chip DDR is the bottleneck. It always is 😉. It is also good news that DDR is being expanded to 128 bits.

The question however was more about the following:
- If the algorithm does not need a Frame Buffer, how adaptive is the _ztachip_ eco-system, both HW and SW, to such "streaming" processing model?!

Here is an interesting excerpt from Xilinx [PG231](https://docs.amd.com/r/en-US/pg231-v-proc-ss/Scaling) on this topic. For the "Scaling Only" mode, they don't need much storage, yet alone a large Frame Buffer. That allows them supporting 
> [image]

That allows them supporting very high [resolutions](https://docs.amd.com/r/en-US/pg231-v-proc-ss/Feature-Summary):
> [image]

ztachip, 2025-06-06:
Please reference
https://github.com/ztachip/ztachip/blob/master/SW/apps/color/kernels/color.m
Example of perform color conversion, processing is streaming by processing in tiles
Also reference section 6 of doc below on procedure to do stream processing
https://github.com/ztachip/ztachip/raw/master/Documentation/ztachip_programmer_guide.pdf

image.png (view on web)<https://github.com/user-attachments/assets/573c6599-4623-4175-ae55-e53b36598788>

image.png (view on web)<https://github.com/user-attachments/assets/e989921e-c56b-4ea7-be83-34ef56965169>